tud-phi/ros2-elastica #10: Implement standard ROS(2) control and sensor msgs

mstoelzle, 2021-10-30:
Currently, you are only subscribing and publishing scalars or arrays of type Float64: https://github.com/tud-cor-sr/ros2-elastica/blob/testing_versions/elastica_sim_control/elastica_sim_control/elastica_publisher_subscriber.py

Our goal is to make the simulator plug-and-play with ROS packages we develop also for lab experiments, but also with other ROS packages developed by the community.

Please implement more standard control, geometry and sensor ROS messages in addition to the current publishing of the raw float messages. Examples for this are PoseArray.msg, AcecelStamped.msg, Vector3.msg etc
https://github.com/ros2/common_interfaces/tree/master/geometry_msgs/msg

For coordinate transformations, tf2 might be quite interesting: http://wiki.ros.org/tf2

RUFFY-369, 2021-11-01:
Hi @mstoelzle , Thanks for pointing this out. I'm currently working on standardising those ROS messages like you pointed out in this [https://github.com/ros2/common_interfaces/tree/master/geometry_msgs/msg](https://github.com/ros2/common_interfaces/tree/master/geometry_msgs/msg).
I'll push the required changes into the repo

RUFFY-369, 2021-11-05:
Hi @mstoelzle , I have added more standard geometry msgs in addition to current publishing of msgs in the form of custom msgs.
Making custom msgs would be helpful, in case, if one wants to add more params or info. Also the code in the `elastica_publisher_subscriber.py` gets reduced as redundant amount of publishers & subscribers are reduced

mstoelzle, 2021-11-05:
Looks very good for now. Maybe when you are actually starting to implement a controller, we might realize that we have to adjust some details.

closed 2021-11-05 by mstoelzle