tud-phi/ros2-elastica #11: Is subscription to control inputs still missing?

mstoelzle, 2021-10-30:
I see that you are subscribing to some topics: https://github.com/tud-cor-sr/ros2-elastica/blob/testing_versions/elastica_sim_control/elastica_sim_control/elastica_publisher_subscriber.py#L270

However, it does not seem like you are doing anything with the received messages except printing them. I am missing the subscription to the control inputs and the forwarding to the elastica simulation.

RUFFY-369, 2021-11-30:
Hi @mstoelzle , As discussed in the last meet for the control input, I have modified the scripts accordingly. 
For now, in the 'elastica_publisher_subscriber.py' script, I am passing random control input (the control points for the beta spline in normal direction) to the elastica simulator.

RUFFY-369, 2021-12-01:
is this issue resolved as the discussion on the control is in a separate one?

closed 2021-12-01 by mstoelzle