tud-phi/ros2-elastica #12: ROS parameters

mstoelzle, 2021-10-30:
Please think about which parameters should be initialized as ROS parameters and put into a launch file.

For now, I am mainly thinking of the timer frequency, which is currently hard-coded: https://github.com/tud-cor-sr/ros2-elastica/blob/testing_versions/elastica_sim_control/elastica_sim_control/elastica_publisher_subscriber.py#L89

RUFFY-369, 2021-11-30:
Hi @mstoelzle , I've added the ROS params the recent commits.

mstoelzle, 2021-11-30:
Please also add the timer period for the callback (e.g. the period for publishing state updates). You might also want to decide to call it `pub_freq` (e.g. publication frequency in the unit Hz - this is how most people usually think about it)

RUFFY-369, 2021-12-01:
Oh! Yeah sure that was the remaining one along with buffer size. 
Usually people say it triggers for every half second so, may be frequency would be better to use here for dev clarity purpose.

RUFFY-369, 2022-01-15:
Hi @mstoelzle , I've solved the issue in the recent commits, please review

closed 2022-01-15 by mstoelzle