tud-phi/ros2-elastica #15: Publishing of ControlInput messages

mstoelzle, 2021-11-30:
I do not understand why you publish the control input message again within the same node. Our controller, should go into a different node and that one should publish suitable control commands as ControlInput messages:

https://github.com/tud-cor-sr/ros2-elastica/blob/main/elastica_sim_control/elastica_sim_control/elastica_publisher_subscriber.py#L123

RUFFY-369, 2021-12-01:
In the main script where both processes are started I initialised the Value object for the initial passing as a control command in time_stepping. The same object I passed to the ros node where it was being published & its respective subscriber changed the control points linked with the simulator.
Regarding what you're saying, do you want a separate node for publishing of just control inputs & from their the subscriber can change the value of control points?

mstoelzle, 2021-12-01:
Yes, we will have separate control node(s) which will publish the control input messages (please refer first to #16).
The 'elastica_sim' package with its nodes for the specific environments should only subscribe to these control input messages, run the simulation and re-publish the states of the robot.

RUFFY-369, 2021-12-01:
Okay so, one node for the simulation parameters & state and an other node for the control, right?
I just asked some questions in #16

mstoelzle, 2021-12-01:
> Okay so, one node for the simulation parameters & state and an other node for the control, right? I just asked some questions in #16

Maybe you just miss-spelled, but I meant one package for simulation and one package for running the controller. The simulation package subscribes to the control input topic and passes the control input commands along to the simulator. At the same time the simulation package publishes the robots state and the controller package will most likely subscribe to this robot state

RUFFY-369, 2021-12-01:
Yeah, just realised that, my bad.
I'll move the controller to a separate package.

RUFFY-369, 2022-01-15:
Hi @mstoelzle , in the recent commits I've pushed the new package for controller. Please review it.

closed 2022-01-16 by mstoelzle