tud-phi/ros2-elastica #16: Re-work actuation model

mstoelzle, 2021-11-30:
After looking at your current implementation and looking through Elasticas tutorials and code, I realised that we need to change the actuation model:

- We consider a soft robotic arm actuated pneumatically. Pneumatic actuation allows us to apply a constant torque over a specified sub-length of the arm (also called a segment).
- We should implement this analogy to pneumatic actuation also in simulation. I think the easiest strategy would be to connect multiple rods with [fixed joints](https://github.com/GazzolaLab/PyElastica/blob/master/elastica/joint.py#L196).  Every rod represents aa segment with one pneumatic actuation model. This means, that we should apply a [uniform torque](https://github.com/GazzolaLab/PyElastica/blob/master/elastica/_elastica_numpy/_external_forces.py#L134) along the rod. The control input in this case should be a sequence of torques for each segment.

- [ ] Please find out into which direction the UniformTorque on a rod is applied to. Is this in the world coordinate frame or the local coordinate frame of the rod? If we cannot adapt the direction of control online, we should add a UniformTorque into each cardinal direction, so that we can scale each cartesian torque component during the simulation

Accordingly, the current ControlInput msg can be kept, but should be renamed to MuscleTorques msg with fields direction, angular_frequency, wave_number, phase_shift, ramp_up_time and spline_control_points as documented here:
https://github.com/GazzolaLab/PyElastica/blob/master/elastica/_elastica_numpy/_external_forces.py#L200

mstoelzle, 2021-11-30:
I see that the torque is multiplied with the director collection, so probably its in a local coordinate frame?
https://github.com/GazzolaLab/PyElastica/blob/master/elastica/_elastica_numpy/_external_forces.py#L162

RUFFY-369, 2021-12-01:
> After looking at your current implementation and looking through Elasticas tutorials and code, I realised that we need to change the actuation model:
>
> * We consider a soft robotic arm actuated [...]

- Yeah in the current actuation model the arm is considered like a single strand of muscle where torque is applied in the form of beta spline for the actuation of whole arm together.
- Regarding changing the actuation model to pneumatically actuated arm how're you thinking of the arrangement of the segments. Is it like (for now) a single linear attachment of segments (which can each be seen as a pneumatic chamber) making the whole arm?
- Should we assume that the torque in each pneumatic segment is uniform throughout the actuation if we're using [uniform torque](https://github.com/GazzolaLab/PyElastica/blob/master/elastica/_elastica_numpy/_external_forces.py#L134)? 
- Also, what values of the properties of material should we take for each segment, i.e., NU(energy dissipation coefficient), poisson ratio & young's modulus. Will it be better to choose those values after defining the environment?

- Yeah I'll find that out.
 
- We're not using the MuscleTorque class as that gives the muscle torque in the form of a travelling wave for which these (angular_frequency, wave_number, phase_shift, ramp_up_time ) params are required but instead just the control points' values.

- Also, can we move the current actuation model where varying beta spline is used for the actuation of the whole arm to a different branch from main. And push the commits for new pneumatic actuation model to the main branch

mstoelzle, 2021-12-01:
- I do not understand your second bullet point fully. I imagine that we would build a chain of Cosserat rods and each rod represents one pneumatically actuated segments. These rods are connected with rigid joints. Each segment contains multiple pneumatic chambers, which we do not model. We only model the resulting, uniform torque by the pneumatic force on the segment. Does this answer your question?
- Yes, we can assume that we have a uniform torque along the pneumatic segment as this torque is proportionally related to the pressure in the chambers.

RUFFY-369, 2021-12-01:
- Yeah I was talking about the abstraction of the number of pneumatic chambers in a single segment & also about their linkage with other such segments, i.e. a serial chain or may some parallel attachments as well?
- Oh! Alright. Then we just need to use the uniform torque class instead of the varying beta spline.
- Can you describe me more about the behavior of the SR arm at the lab to get some approx model in the simulation?
- Also, what should be the number of elements of each pneumatic segment/rod (current simulation has 50 elements for the whole arm)? The more the elements the more flexible the segment will be. I'm thinking of making it 1/10th but that can be tested out & decided. If you have some other number in mind, then please share

mstoelzle, 2021-12-01:
- We do not model the pneumatic chambers directly. Thus, we only have a serial linkage between segments and no parallel attachments. The behaviour of 4 chambers is all represented in the **one** UniformTorque we are applying on the segment.
- The robot should behave similarily as in this video: https://www.youtube.com/watch?v=AFknWDTHncM These robots have elasticity in the sense that if no actuation or external forces are acting on them, they will bend-back to a straight configuration
- Is the number of elements per Cosserat rod (e.g. per segment) not mainly a parameter to trade-off computational time with simulation accuracy? I think it should not have a major influence on the flexibility of the segment. Yes, this will be something to tune manually depending how the behaviour looks like

RUFFY-369, 2021-12-01:
- Okay, thanks for the clarification.
- Alright, will ensure that. Okay so, kind of like a restoring force which is always acting in the absence as well the presence of external torques, right?
- No, I don't think it is a param for a trade-off between both of them. In the docs they have mentioned that if we want our rod/segment to be more flexible then increase the number of elements. Doubling the number of elements do a 10-20% increase in run time.

RUFFY-369, 2021-12-01:
> * Yeah in the current actuation model the arm is considered like a single strand of muscle where torque is applied in the form of beta spline for the actuation of whole arm together.
> * Regarding [...]

This in reference to the last bullet point question for changing the branch name

mstoelzle, 2021-12-01:
> * Alright, will ensure that. Okay so, kind of like a restoring force which is always acting in the absence as well the presence of external torques, right?

Yes, exactly. Possibly, this is already a property of cosserad rods? If not, we need to add it manually. The force is always proportional to the curvature of the segment:
```
tau_elastic = K @ kappa
```

RUFFY-369, 2021-12-01:
> Yes, exactly. Possibly, this is already a property of cosserad rods? If not, we need to add it manually. The force is always proportional to the curvature of the segment:
>
> ```
> tau_elastic = K @ [...]

Yes, @mstoelzle  it is already there in elastica. I just simulated the flagella structure & visualised to confirm it. For first 3 seconds one set of control points, for another three different set & then in the remaining time control points are set to a zero array. The behaviour can be seen in the GIF below:
[image]

mstoelzle, 2021-12-01:
> Yes, @mstoelzle it is already there in elastica. I just simulated the flagella structure & visualised to confirm it. For first 3 seconds one set of control points, for another three different set & [...]

Ok, nice! So this is not something we have to specifically address for now then

RUFFY-369, 2021-12-01:
Yep this satisfies one behavioural characteristic  now for the simulation of the SR arm you mentioned

RUFFY-369, 2022-01-15:
Hi @mstoelzle , In the recent commits I've changed the actuation model to the following:
**Number of rods** (pneumatic segments): 6
**Number of elements per rod** : 50
**Control Method**: Uniform Torque in three directions per segment
**Joint between segments:** Fixed joint

closed 2022-01-15 by mstoelzle