tud-phi/ros2-elastica #17: Avoid copying of Elastica code

mstoelzle, 2021-11-30:
Please try to avoid copying the elastica code and instead try to find another way for the implementation such as class inheritance etc

https://github.com/tud-cor-sr/ros2-elastica/blob/main/elastica_sim_control/elastica_sim_control/muscle_torques_with_bspline_numba.py
https://github.com/tud-cor-sr/ros2-elastica/blob/main/elastica_sim_control/elastica_sim_control/muscle_torques_with_bspline_numpy.py

RUFFY-369, 2021-12-01:
Usually all the force classes are inherited from NoForces class. We can define our own Force/Torque class to provide the actuation. In this [class](https://github.com/tud-cor-sr/ros2-elastica/blob/main/elastica_sim_control/elastica_sim_control/muscle_torques_with_bspline_numba.py) they coded for varying beta spline if I make a separate class then the working & code would be similar to [this](https://github.com/tud-cor-sr/ros2-elastica/blob/main/elastica_sim_control/elastica_sim_control/muscle_torques_with_bspline_numba.py).
And for inheritance we can use the classes which are availble in the elastica package i.e.,   NoForces, GravityForces, EndpointForces, UniformTorques, UniformForces, MuscleTorques. And only NoForce is needed & useful for inheritance.

So, do you want me to make a custom class like [this](https://github.com/tud-cor-sr/ros2-elastica/blob/main/elastica_sim_control/elastica_sim_control/muscle_torques_with_bspline_numba.py) for passing it add_forcing() method & applying actuation with changing control points?

mstoelzle, 2021-12-01:
Please refer first to issue #16.

Let me put the priorities like this:
1. If possible, please use the umodified force classes. Then, you can just import them from the elastica package without having to copy them into our repo
2. If you need to modify a class, please inherit from the most similar class. You could do something like this:
```
from elastica.x.external_forces import UniformTorques

class CustomUniformTorques(UniformTorques):
    def __init__(self, torque, direction=np.array([0.0, 0.0, 0.0])):
        super(CustomUniformTorques, self).__init__()
        
        INSERT_YOUR_CUSTOM_CODE

    def apply_torques(self, system, time: np.float64 = 0.0):
        super(CustomUniformTorques, self).apply_torques(system, time)

        INSERT_YOUR_CUSTOM_CODE
```

RUFFY-369, 2021-12-01:
So, for now as we are changing the actuation model do I need to customise the varying beta spline class instead of using their module from Elastica-RL or will it be better to move on with creating the custom class for uniform torques where the applied torque keeps changing according to the controller

mstoelzle, 2021-12-01:
> So, for now as we are changing the actuation model do I need to customise the varying beta spline class instead of using their module from Elastica-RL or will it be better to move on with creating [...]

You can directly move on to the UniformTorques and the revised actuation model

RUFFY-369, 2021-12-01:
Okay sure

RUFFY-369, 2022-01-15:
Hi @mstoelzle , in the new actuation model, I've used the unmodified force class of Uniform torque. Kindly test it

closed 2022-01-15 by mstoelzle