tud-phi/ros2-elastica #18: Better naming of subscribers and publisher

mstoelzle, 2021-11-30:
Please use a more intuitive name for the publisher and subscriber class attributes:
https://github.com/tud-cor-sr/ros2-elastica/blob/main/elastica_sim_control/elastica_sim_control/elastica_publisher_subscriber.py#L41

RUFFY-369, 2021-12-01:
Yeah sure, those were needed to be changed from just numbers.
Also, are the ros topic names fine?

mstoelzle, 2021-12-01:
The topic names should also be ROS parameters specified in the config. The default value should be in the elastica namespace `/elastica/physical_params`, `/elastica/rod_state`, `/elastica/muscle_torque_control_points` and `/elastica/uniform_torques`

RUFFY-369, 2021-12-01:
Okay that will be a more standard way for naming the ros topics. Separate params for each package naming their respective ros topics

RUFFY-369, 2022-01-15:
Hi @mstoelzle , Solved in recent commits

mstoelzle, 2022-01-15:
I think what is still missing is to specify the publication and subscription topic names through ROS parameters. And the default topic name (e.g. if not specified otherwise in the ROS parameters) should be in the `elastica` name space. This means that you need to remove the `/` at the beginning of the default topic name:

https://github.com/tud-cor-sr/ros2-elastica/blob/main/elastica_sim/elastica_sim/elastica_publisher_subscriber.py#L37

RUFFY-369, 2022-01-15:
Oh! My bad, i missed changing the namespace for the ros topics. I will make the necessary changes.

RUFFY-369, 2022-01-16:
Hi @mstoelzle , I pushed the new topic names in config in the recent commits. Please review them

closed 2022-01-16 by mstoelzle