tud-phi/ros2-elastica #19: Is multiprocessing really necessary?

mstoelzle, 2021-11-30:
I am wondering if the multiprocessing with one process each for simulation and ROS node is really necessary? I understand that the reason that you use the multiprocessing.Array is to share information between both processes?

Would it not be possible to run both the ROS node and the simulator within the same process? In the `ContinuumFlagellaCallBack` class you would initiate the publishing of the current state of the robot (in the same process as the simulator) and then obviously with `time_stepping` we would communicate the new control input which we have received from the ROS subscriber

I just want to make sure that the implemented solution is not more complicated than necessary, before we adapt to code to more environments and simulations.

RUFFY-369, 2021-12-01:
- Regarding the publishing in `ContinuumFlagellaCallBack`: before that we have to spin the declared ros node & when that happens other work will be blocked until it is terminated and then the code control won't return from the callback function. (Check the last point where I tried to resolve this blockage issue of ros node)
- Before using the time_stepping function in the DefineFlagella class, I was using the integrate() function of elastica module but whenever we call it, the code control goes into the loop of performing the required number of time steps. So, I started using Multiprocessing to separate the processes.
- After dropping the integrate function & moving to make the time_stepping function,  I tried passing the flagella_sim object to our publisher subscriber node  & then perform time stepping in the timer callback function but it was way too slow. Check the GIF down below:

[image]

**So, these were the reasons I sticked with using multiprocessing with two processes of simulation & spinning the ros node**

RUFFY-369, 2021-12-01:
Is this issue resolved?

mstoelzle, 2021-12-01:
Yes, I will close this issue for now. It will be available anyway for future reference

closed 2021-12-01 by mstoelzle