tud-phi/ros2-elastica #23: Fix RodState msg type

mstoelzle, 2022-01-15:
I am a bit confused that here
https://github.com/tud-cor-sr/ros2-elastica/blob/main/elastica_sim/elastica_sim/continuum_flagella_ros2.py#L47
you say that there are 51 rods, but then, you only publish only the state of 6 rods here:
https://github.com/tud-cor-sr/ros2-elastica/blob/main/elastica_msgs/msg/RodState.msg

I find the structure of the `RodState.msg` way too inflexible. I propose the following structure:

In the `RodStates.msg`, you report the state of all rods (however many the user chose to use for the simulation). Accordingly, the message is structured like:
```
std_msgs/Header header 
# number of rods used in simulation
int num_rods

# Actual poses of the rods as an array of dimension num_rods
geometry_msgs/PoseStamped[] poses
```

It's always the better choice to use a standard message format such as `PoseStamped` whenever possible. Also, this gives us the flexible to use however many rods we want to use

RUFFY-369, 2022-01-16:
Hi @mstoelzle , yeah the rod state msg structure was too inflexible for the time being and also too loosen. I have the refactored the code and structured new one. I was delaying to use the above approach for the future code after changing the actuation model.
From now on I will maintain the standard messages in the new message structure beforehand. Thanks
I've pushed the required  changes in the recent commits. Please review them.

mstoelzle, 2022-01-16:
I think it looks good in general. Two small changes I would do for consistency:

1. Rename `velocity` to `velocities` in `RodState.msg`: https://github.com/tud-cor-sr/ros2-elastica/blob/main/elastica_msgs/msg/RodState.msg#L10
2. Rename `rods_state` to `rod_states` in `RodsState.msg`: https://github.com/tud-cor-sr/ros2-elastica/blob/main/elastica_msgs/msg/RodsState.msg#L9

RUFFY-369, 2022-01-17:
Hi @mstoelzle , pushed the minor changes.

closed 2022-01-17 by mstoelzle