tud-phi/ros2-elastica #25: Elastica Kinematics

mstoelzle, 2022-01-16:
I am creating this issue for future reference. Please implement first a soft robotic arm and a sphere into the simulation as already discussed previously. 

Afterwards, I think the next step is to add some kinematics. This will be quite a bit more mathematical, than all the coding in the first part of the internship :) I am thinking of the following structure:

- The package should be named `elastica_kinematics`
- A main component of (inverse) kinematics is usually the Jacobian. You will need to derive this symbolically. However, it is very computationally inefficient to do this at run-time. Therefore, I propose to add a script `elastica_kinematics/scripts/derive_jacobian.py`, which computes the Jacobian symbolically using [sympy](https://www.sympy.org/en/index.html) and afterwards saves the result as a compiled C++ library. Then, our ROS nodes can access this C++ library for fast usage at run-time. We should have a Jacobian available at the minimum for every tip of each segment. Ideally, we can also evaluate the Jacobian within each segment as a function of a running variable `s` with minimum 0 (e.g. the base of the segment) and 1 as the maximum (e.g. the tip of the segment).
- We have these rod states now. These are way to high-dimensional though to use them for any kind of feedback control purposes. Additionally, we want to re-use as much of the learnings from literature as possible. Therefore, we should derive a standard Piecewise Constant Curvature (PCC) kinematic description of the soft robotic state. This means that we model every segment to have constant curvature, even if there are small deviations in the rod states of one segment. The node could have the name `RodStatesToPCC` with the `RodsState.msg` as an input and the PCC kinematic states as an output. We aim to use the following PCC convention:
```
Della Santina, Cosimo, Antonio Bicchi, and Daniela Rus. "On an improved state parametrization for soft robots with piecewise constant curvature and its use in model based control." IEEE Robotics and Automation Letters 5.2 (2020): 1001-1008.
```
- Please implement a `ForwardKinematics` node, which subscribes to a topic with PCC configuration states and publishes the position and orientation of the tip of each segment.

RUFFY-369, 2022-02-24:
Hi @mstoelzle , please review the latest commits regarding this issue

mstoelzle, 2022-03-01:
> Hi @mstoelzle , please review the latest commits regarding this issue

I have created the following issues to address specific concerns with the current implementation:

- [ ] #26 
- [ ] #27 
- [ ] #28
- [ ] #29

RUFFY-369, 2022-03-06:
> > Hi @mstoelzle , please review the latest commits regarding this issue
> 
> I have created the following issues to address specific concerns with the current implementation:
> 
> * [ ]  [Remove no_of_objects variable #26](https://github.com/tud-cor-sr/ros2-elastica/issues/26)
> * [ ]  [Rename no_of_segments to num_segments and add as ROS2 parameter #27](https://github.com/tud-cor-sr/ros2-elastica/issues/27)
> * [ ]  [Use improved PCC paramterization #28](https://github.com/tud-cor-sr/ros2-elastica/issues/28)
> * [ ]  [Double-check spelling of normal_director #29](https://github.com/tud-cor-sr/ros2-elastica/issues/29)
> * [ ]  [Strategy for closed-form inverse kinematics #30](https://github.com/tud-cor-sr/ros2-elastica/issues/30)

Sure, will go through them asap to solve