tud-phi/ros2-elastica #26: Remove no_of_objects variable

mstoelzle, 2022-03-01:
I do not see this variable used anywhere in the code

https://github.com/tud-cor-sr/ros2-elastica/blob/436afc30f7909b9397da65641e277c39e012fb61/elastica_kinematics/elastica_kinematics/elastica_rod_states_to_PCC.py#L31

RUFFY-369, 2022-03-06:
Hi @mstoelzle , this variable is for future use when we have more than one number of objects, for that we need to code the same function as creating multiple rods/segments

mstoelzle, 2022-03-07:
> Hi @mstoelzle , this variable is for future use when we have more than one number of objects, for that we need to code the same function as creating multiple rods/segments

@RUFFY-369 Can you please explain me what is the difference between a _segment_ and an _object_?

RUFFY-369, 2022-03-08:
@mstoelzle , objects are the one's like spheres which we will try to manipulate in the sim and segments are our pneumatic model segments which are joined with fixed_joint making up the arm

mstoelzle, 2022-03-08:
> @mstoelzle , objects are the one's like spheres which we will try to manipulate in the sim and segments are our pneumatic model segments which are joined with fixed_joint making up the arm

Ok, I understand now. But then these external objects should not be introduced in the kinematics package, as we do not need to compute any kinematics for these external objects (as least what we know so far).

RUFFY-369, 2022-03-13:
yeah but their positions and orientations need to be used by the controller so, I need to  shift them in the sim pack and publish the states there

mstoelzle, 2022-03-14:
> yeah but their positions and orientations need to be used by the controller so, I need to shift them in the sim pack and publish the states there

I would say that positions and orientations of external objects would need to used by a planner (e.g. planning the trajectory of the soft robot), and then the (more low-level) controller actually guides the robot towards this configuration. In any case, this means that the external object would probably not need to be represented in the kinematics package, but more likely in a future planner package

RUFFY-369, 2022-03-16:
Oh! okay like the config of the objects for the trajectory or path planner and then our motion planner will guide the actuation for those trajectories.