tud-phi/ros2-elastica #29: Double-check spelling of normal_director

mstoelzle, 2022-03-01:
https://github.com/tud-cor-sr/ros2-elastica/blob/436afc30f7909b9397da65641e277c39e012fb61/elastica_msgs/msg/RodState.msg#L19

What is the meaning of the `normal_director`? is it the direction of the local `z-axis` in the global reference frame? If it is, then we should probably call it differently. Is `director` even a word in the dictionary?

RUFFY-369, 2022-03-06:
Basically, in pyelastica they have this 3X3 matrix for each element which consists of a triad of orthonormal basis vectors and in their code they named the vectors as directors (tangent, normal and binormal). So, normal_director is the normal vector or you can assign it as z-axis. To be consistent with code of pyelastica I assigned it as normal_director_base for the base of the rod

mstoelzle, 2022-03-08:
> Basically, in pyelastica they have this 3X3 matrix for each element which consists of a triad of orthonormal basis vectors and in their code they named the vectors as directors (tangent, normal and [...]

Would it make sense to save / publish all three directors (so tangent, normal and binormal) instead of just the normal director?

RUFFY-369, 2022-03-13:
as the directors are basically can be used to get the orientation in the form of quaternions so the quaternions are already being published. normal_director is being used to get the change in orientation of the normal vector between the base and the tip