tud-phi/ros2-elastica #30: Strategy for closed-form inverse kinematics

mstoelzle, 2022-03-01:
https://github.com/tud-cor-sr/ros2-elastica/blob/436afc30f7909b9397da65641e277c39e012fb61/elastica_kinematics/elastica_kinematics/elastica_rod_states_to_PCC.py#L93

I am not convinced of fully relying on the rotation between the base and the tip to estimate the configuration of the segment. This strategy, can for example not capture any elongation of the segment. I would prefer to implement it analogue to what we are currently doing experimentally in our robot, which also takes into account the translation between the base and the tip.

You can use the proposed strategy in Equation (17) of the improved state parametrization paper by Cosimo and the attached screenshot. 
[image]

RUFFY-369, 2022-03-06:
In the following:
[L50](https://github.com/tud-cor-sr/ros2-elastica/blob/436afc30f7909b9397da65641e277c39e012fb61/elastica_kinematics/elastica_kinematics/elastica_rod_states_to_PCC.py#L50), I've used the 4X4 matrix consisting of both the translation as well as rotation for configuration estimation of the consecutive segments.
Alright, I'll try to implement the above mentioned strategy

mstoelzle, 2022-03-07:
> 

As I understand the code, the `transformation` method  in 
https://github.com/tud-cor-sr/ros2-elastica/blob/436afc30f7909b9397da65641e277c39e012fb61/elastica_kinematics/elastica_kinematics/elastica_rod_states_to_PCC.py#L50 
is doing forward kinematics (so using the known configuration to compute the transformation). This GitHub issue is referring to using both the translation and rotation for inverse kinematics (so using the transformation to estimate a configuration)

RUFFY-369, 2022-03-13:
but I think we planned to use the jacobian script for the inverse kinematics right?

RUFFY-369, 2022-03-16:
Yeah the first which we have right now which is to be modified as you mentioned in the issue and for the second, not immediate one, the Jacobian script has already been made which can be changed later on accordingly